riscv/riscv-unified-db #731: Still have 404 errors on artifacts web page even after nightly passes

james-ball-qualcomm, 2025-05-06:
The nightly regression passed for the 1st time in 3 weeks last night. I was hoping all the 404 artifact link errors on https://riscv-software-src.github.io/riscv-unified-db/index.html would have been fixed. However, the following links still are 404:

- All profiles
- ISA Explorer Excel (https://riscv-software-src.github.io/riscv-unified-db/isa_explorer.xlsx)
- AC100 and AC200 CRDs
- IDL Language doc (https://riscv-software-src.github.io/riscv-unified-db/ruby/idl/index.html)
- Ruby Interface doc (https://riscv-software-src.github.io/riscv-unified-db/ruby/arch_def/index.html)

AFOliveira, 2025-05-06:
I have not taken a look at this, but may it be because our deployments are broken and the web view you linked does not represent the most up-to-date version of UDB?

james-ball-qualcomm, 2025-05-06:
Here's a link to last night's regression:
   https://github.com/riscv-software-src/riscv-unified-db/actions/runs/14850973611

I'll start looking through the logs since I've improved the display of information and checking of exit status in PRs now in main. Maybe some of you can do the same in case you spot something useful.

The only failure I see is the "Check Latest Commit" annotation when you can find from the link above. Who knows about this? It says:

[image]

james-ball-qualcomm, 2025-05-06:
Hold on, just found this clue at the bottom of the log output:
 This step has been truncated due to its large size. Download the full logs from the menu once the workflow run has completed.

james-ball-qualcomm, 2025-05-06:
[logs_38132817721.zip](https://github.com/user-attachments/files/20062377/logs_38132817721.zip)

Attached logs.

james-ball-qualcomm, 2025-05-06:
Just realized I'm looking at the test:nightly results when I should be looking at the Deploy Pages action since that is what runs the lib/deploy.sh script and creates the artifacts on https://riscv-software-src.github.io/riscv-unified-db/index.html

james-ball-qualcomm, 2025-05-06:
The Deploy Pages doesn't seem happy even though it built find in my branch 597-isa-explorer-backend-missing-extension-dependencies yesterday:

[image]

AFOliveira, 2025-05-06:
@james-ball-qualcomm maybe you missed my comments, but I suppose that the problem is precisely that, the deploys page is not updated

james-ball-qualcomm, 2025-05-06:
Thanks Afonso. Didn't see your earlier comments. I'm now digging into the actions below that happen after lib/deploy.sh since it seems to be working properly now (was broken for a long time).

https://github.com/riscv-software-src/riscv-unified-db/actions/runs/14846099662/workflow

[image]

james-ball-qualcomm, 2025-05-06:
Looks like the actions above are standard GitHub actions:

- https://github.com/actions/configure-pages
- https://github.com/actions/upload-pages-artifact
- https://github.com/actions/deploy-pages

They probably have some way that they were setup (presumably by Derek) so maybe there's some bit rot there.

james-ball-qualcomm, 2025-05-06:
Actually, I see now that the deploy_pages.sh script is exiting with a non-zero status. This is my first time using the new logging I put into the deploy_pages.sh script to debug failures so you are all watching a learning curve.

james-ball-qualcomm, 2025-05-06:
Looking for "DEPLOY.*failed" in the deploy.sh log, I found the culprit:

```
2025-05-05T21:07:55.5035433Z /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task_manager.rb:59:in `[]': Don't know how to build task 'gen:html_manual MANUAL_NAME=isa VERSIONS=all' (See the list of available tasks with `do --tasks`) (RuntimeError)
2025-05-05T21:07:55.5036749Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:187:in `invoke_task'
2025-05-05T21:07:55.5037703Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block (2 levels) in top_level'
2025-05-05T21:07:55.5038951Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `each'
2025-05-05T21:07:55.5039841Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block in top_level'
2025-05-05T21:07:55.5040739Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:147:in `run_with_threads'
2025-05-05T21:07:55.5041596Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:132:in `top_level'
2025-05-05T21:07:55.5042138Z 	from -e:1:in `<main>'
2025-05-05T21:07:55.5068715Z Running with 1 job(s)
2025-05-05T21:07:55.5251576Z [DEPLOY] Mon May  5 21:07:55 UTC 2025 ./do gen:html_manual MANUAL_NAME=isa VERSIONS=all failed

james-ball-qualcomm, 2025-05-06:
Here's the full run of the ISA manual generation. The command looks good to me. @dhower-qc , any clues?

```
2025-05-05T21:07:53.5166161Z [DEPLOY] Mon May  5 21:07:53 UTC 2025 gen:html_manual MANUAL_NAME=isa VERSIONS=all
2025-05-05T21:07:53.5269890Z Using singularity environment
2025-05-05T21:07:53.5270522Z Running in a GitHub Action
2025-05-05T21:07:53.5415963Z * Applying /usr/lib/sysctl.d/10-apparmor.conf ...
2025-05-05T21:07:53.5416381Z * Applying /etc/sysctl.d/10-bufferbloat.conf ...
2025-05-05T21:07:53.5416942Z * Applying /etc/sysctl.d/10-console-messages.conf ...
2025-05-05T21:07:53.5417520Z * Applying /etc/sysctl.d/10-ipv6-privacy.conf ...
2025-05-05T21:07:53.5418075Z * Applying /etc/sysctl.d/10-kernel-hardening.conf ...
2025-05-05T21:07:53.5418619Z * Applying /etc/sysctl.d/10-magic-sysrq.conf ...
2025-05-05T21:07:53.5419136Z * Applying /etc/sysctl.d/10-map-count.conf ...
2025-05-05T21:07:53.5419674Z * Applying /etc/sysctl.d/10-network-security.conf ...
2025-05-05T21:07:53.5420193Z * Applying /etc/sysctl.d/10-ptrace.conf ...
2025-05-05T21:07:53.5420666Z * Applying /etc/sysctl.d/10-zeropage.conf ...
2025-05-05T21:07:53.5421169Z * Applying /usr/lib/sysctl.d/50-coredump.conf ...
2025-05-05T21:07:53.5421677Z * Applying /usr/lib/sysctl.d/50-pid-max.conf ...
2025-05-05T21:07:53.5422191Z * Applying /etc/sysctl.d/99-cloudimg-ipv6.conf ...
2025-05-05T21:07:53.5422891Z * Applying /etc/sysctl.d/99-cloudimg-udp.conf ...
2025-05-05T21:07:53.5423432Z * Applying /usr/lib/sysctl.d/99-protect-links.conf ...
2025-05-05T21:07:53.5423949Z * Applying /etc/sysctl.d/99-sysctl.conf ...
2025-05-05T21:07:53.5424412Z * Applying /etc/sysctl.d/99-userns.conf ...
2025-05-05T21:07:53.5424842Z * Applying /etc/sysctl.conf ...
2025-05-05T21:07:53.5425275Z kernel.apparmor_restrict_unprivileged_userns = 1
2025-05-05T21:07:53.5425768Z net.core.default_qdisc = fq_codel
2025-05-05T21:07:53.5426005Z kernel.printk = 4 4 1 7
2025-05-05T21:07:53.5426223Z net.ipv6.conf.all.use_tempaddr = 2
2025-05-05T21:07:53.5426493Z net.ipv6.conf.default.use_tempaddr = 2
2025-05-05T21:07:53.5426744Z kernel.kptr_restrict = 1
2025-05-05T21:07:53.5426961Z kernel.sysrq = 176
2025-05-05T21:07:53.5427395Z vm.max_map_count = 1048576
2025-05-05T21:07:53.5427623Z net.ipv4.conf.default.rp_filter = 2
2025-05-05T21:07:53.5427867Z net.ipv4.conf.all.rp_filter = 2
2025-05-05T21:07:53.5428093Z kernel.yama.ptrace_scope = 1
2025-05-05T21:07:53.5428317Z vm.mmap_min_addr = 65536
2025-05-05T21:07:53.5428690Z kernel.core_pattern = |/usr/lib/systemd/systemd-coredump %P %u %g %s %t 9223372036854775808 %h
2025-05-05T21:07:53.5429101Z kernel.core_pipe_limit = 16
2025-05-05T21:07:53.5429312Z fs.suid_dumpable = 2
2025-05-05T21:07:53.5429524Z kernel.pid_max = 4194304
2025-05-05T21:07:53.5429737Z net.ipv6.conf.all.use_tempaddr = 0
2025-05-05T21:07:53.5429985Z net.ipv6.conf.default.use_tempaddr = 0
2025-05-05T21:07:53.5430239Z net.core.rmem_max = 1048576
2025-05-05T21:07:53.5430449Z net.core.rmem_default = 1048576
2025-05-05T21:07:53.5430663Z fs.protected_fifos = 1
2025-05-05T21:07:53.5430865Z fs.protected_hardlinks = 1
2025-05-05T21:07:53.5431067Z fs.protected_regular = 2
2025-05-05T21:07:53.5431266Z fs.protected_symlinks = 1
2025-05-05T21:07:53.5431474Z vm.max_map_count = 262144
2025-05-05T21:07:53.5431680Z fs.inotify.max_user_watches = 655360
2025-05-05T21:07:53.5431922Z fs.inotify.max_user_instances = 1280
2025-05-05T21:07:53.5432145Z vm.mmap_rnd_bits = 28
2025-05-05T21:07:53.5432568Z kernel.apparmor_restrict_unprivileged_userns = 0
2025-05-05T21:07:53.5432862Z vm.max_map_count = 262144
2025-05-05T21:07:53.5433072Z fs.inotify.max_user_watches = 655360
2025-05-05T21:07:53.5433307Z fs.inotify.max_user_instances = 1280
2025-05-05T21:07:53.5433530Z vm.mmap_rnd_bits = 28
2025-05-05T21:07:55.5035433Z /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task_manager.rb:59:in `[]': Don't know how to build task 'gen:html_manual MANUAL_NAME=isa VERSIONS=all' (See the list of available tasks with `do --tasks`) (RuntimeError)
2025-05-05T21:07:55.5036749Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:187:in `invoke_task'
2025-05-05T21:07:55.5037703Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block (2 levels) in top_level'
2025-05-05T21:07:55.5038951Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `each'
2025-05-05T21:07:55.5039841Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block in top_level'
2025-05-05T21:07:55.5040739Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:147:in `run_with_threads'
2025-05-05T21:07:55.5041596Z 	from /home/runner/work/riscv-unified-db/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:132:in `top_level'
2025-05-05T21:07:55.5042138Z 	from -e:1:in `<main>'
2025-05-05T21:07:55.5068715Z Running with 1 job(s)
2025-05-05T21:07:55.5251576Z [DEPLOY] Mon May  5 21:07:55 UTC 2025 ./do gen:html_manual MANUAL_NAME=isa VERSIONS=all failed

james-ball-qualcomm, 2025-05-07:
I've beefed up deploy.sh to make it clearer what happened and what failed. Here's an example of what it looks like when I pass in 3 bogus "do" targets:

```
[DEPLOY] Wed May  7 09:56:46 UTC 2025 ***************************************************************
[DEPLOY] Wed May  7 09:56:46 UTC 2025 *                      DEPLOY STARTING                        *
[DEPLOY] Wed May  7 09:56:46 UTC 2025 ***************************************************************
[DEPLOY] Wed May  7 09:56:46 UTC 2025 
[DEPLOY] Wed May  7 09:56:46 UTC 2025 vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
[DEPLOY] Wed May  7 09:56:46 UTC 2025 ./do abc
[DEPLOY] Wed May  7 09:56:46 UTC 2025 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[DEPLOY] Wed May  7 09:56:46 UTC 2025 
Using devcontainer environment
/workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task_manager.rb:59:in `[]': Don't know how to build task 'abc' (See the list of available tasks with `do --tasks`) (RuntimeError)
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:187:in `invoke_task'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block (2 levels) in top_level'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `each'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block in top_level'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:147:in `run_with_threads'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:132:in `top_level'
	from -e:1:in `<main>'
Running with 1 job(s)
[DEPLOY] FAIL Wed May  7 09:56:47 UTC 2025 ./do abc
[DEPLOY] Wed May  7 09:56:47 UTC 2025 
[DEPLOY] Wed May  7 09:56:47 UTC 2025 vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
[DEPLOY] Wed May  7 09:56:47 UTC 2025 ./do def
[DEPLOY] Wed May  7 09:56:47 UTC 2025 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[DEPLOY] Wed May  7 09:56:47 UTC 2025 
Using devcontainer environment
/workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task_manager.rb:59:in `[]': Don't know how to build task 'def' (See the list of available tasks with `do --tasks`) (RuntimeError)
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:187:in `invoke_task'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block (2 levels) in top_level'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `each'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block in top_level'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:147:in `run_with_threads'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:132:in `top_level'
	from -e:1:in `<main>'
Running with 1 job(s)
[DEPLOY] FAIL Wed May  7 09:56:49 UTC 2025 ./do def
[DEPLOY] Wed May  7 09:56:49 UTC 2025 
[DEPLOY] Wed May  7 09:56:49 UTC 2025 vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
[DEPLOY] Wed May  7 09:56:49 UTC 2025 ./do ghi jkl
[DEPLOY] Wed May  7 09:56:49 UTC 2025 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[DEPLOY] Wed May  7 09:56:49 UTC 2025 
Using devcontainer environment
/workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/task_manager.rb:59:in `[]': Don't know how to build task 'ghi' (See the list of available tasks with `do --tasks`) (RuntimeError)
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:187:in `invoke_task'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block (2 levels) in top_level'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `each'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:138:in `block in top_level'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:147:in `run_with_threads'
	from /workspaces/riscv-unified-db/.home/.gems/ruby/3.2.0/gems/rake-13.2.1/lib/rake/application.rb:132:in `top_level'
	from -e:1:in `<main>'
Running with 1 job(s)
[DEPLOY] FAIL Wed May  7 09:56:50 UTC 2025 ./do ghi jkl
[DEPLOY] Wed May  7 09:56:50 UTC 2025 
[DEPLOY] Wed May  7 09:56:50 UTC 2025 ***************************************************************
[DEPLOY] Wed May  7 09:56:50 UTC 2025 *                      DEPLOY FAILED                          *
[DEPLOY] Wed May  7 09:56:50 UTC 2025 ***************************************************************
[DEPLOY] Wed May  7 09:56:50 UTC 2025 
[DEPLOY] Wed May  7 09:56:50 UTC 2025 LIST OF FAILURES:
[DEPLOY] Wed May  7 09:56:50 UTC 2025   ./do abc
[DEPLOY] Wed May  7 09:56:50 UTC 2025   ./do def
[DEPLOY] Wed May  7 09:56:50 UTC 2025   ./do ghi jkl
[DEPLOY] Wed May  7 09:56:50 UTC 2025 
[DEPLOY] Wed May  7 09:56:50 UTC 2025 COMPLETE - Overall exit status is 1

james-ball-qualcomm, 2025-05-07:
Just found a nasty copy&paste bug in deploy.sh that I think is responsible for the Profile artifacts not showing up.

james-ball-qualcomm, 2025-05-07:
Also just fixed a bug in deploy.sh with wrong link to ISA explorer spreadsheet.

james-ball-qualcomm, 2025-05-07:
And bad link in deploy.sh to profile releases.

closed 2025-05-20 by james-ball-qualcomm