riscv/riscv-unified-db #744: Inconsistent grammar in instruction description yaml files

kbroch-rivosinc, 2025-05-07:
## Problem

Currently instruction descriptions are being written inconsistently.

Most notably some say ["This instruction..."](https://github.com/riscv-software-src/riscv-unified-db/blob/2f061dfa14d507487a17ceca8c390d607888f75b/arch/inst/Zbs/binvi.yaml#L8) while others simple start with a verb like ["add an immediate ..."](https://github.com/riscv-software-src/riscv-unified-db/blob/2f061dfa14d507487a17ceca8c390d607888f75b/arch/inst/I/addi.yaml#L7)

## Proposed solution

Description should NOT start with "This instruction".  That should be implicit.
If we really want it added this can be done in the template.  Also this gives us the option to do something like 
put the `inst.long_name` `inst.description` as a sentence instead which I've seen in other inst/pg appendices

ThinkOpenly, 2025-05-08:
This is certainly important for consistency.

I expect all of the prosaic content to be in some churn for some time as we work to generate more and more spec content from UDB. And changes to this content very likely need to go through a ratification process at some point. These types of consistency improvements should be done before that, though. (Unless we manage to generate bit-for-bit exact match to the then-current spec contents, which I doubt we can do because of the inconsistencies already in the spec itself.)

TL;DR: I wouldn't spend a lot of time on this at present, IMHO.

ThinkOpenly, 2025-05-08:
Also note that some of the instruction "description" attributes are quite long (ex. `cbo.inval.yaml` including paragraphs, lists, tables, all of which might unduly clutter the display).

kbroch-rivosinc, 2025-05-12:
Mostly wanted to see if we agreed there's a universal consistent rule for the grammar of an `inst.description` (at least the first sentence). If so we could at least clearly define it and then at first it could be manually enforced at code review.  Later it can hopefully be part of a vale linter rule similar to this discussion: https://github.com/riscv-software-src/riscv-unified-db/issues/293

As for fixing some of the existing issues (if my implicit "This instruction" applies). 

Here's existing scope of problem (NOTE: This doesn't detect where `inst.longname` is at start of `inst.description`):

```
kbroch@kbroch-mac.local:~/rvi/non-repo-repos/riscv-unified-db on  main via  v22.14.0 via 🐍 v3.13.3 via 💎 v2.6.10 took 7s
❯ for f in arch/inst/*/*.yaml; do if [ "$(yq '.description' "$f" | grep -c '^This instruction')" -gt 0 ]; then echo "$f"; fi; done

arch/inst/C/c.addi4spn.yaml
arch/inst/Zawrs/wrs.nto.yaml
arch/inst/Zba/slli.uw.yaml
arch/inst/Zcb/c.not.yaml
arch/inst/Zcb/c.sext.b.yaml
arch/inst/Zcb/c.sext.h.yaml
arch/inst/Zcb/c.zext.b.yaml
arch/inst/Zcb/c.zext.h.yaml
arch/inst/Zcmp/cm.pop.yaml
arch/inst/Zcmp/cm.popret.yaml
arch/inst/Zcmp/cm.popretz.yaml
arch/inst/Zcmp/cm.push.yaml
arch/inst/Zicond/czero.eqz.yaml
arch/inst/Zicond/czero.nez.yaml
```

ThinkOpenly, 2025-05-13:
I'm not against this idea, nor consistency in general. I suggest we pose this to the sig-documentation list. I'd consider them more appropriate owners for language-specifics. Ideally, the text in the spec is changed to match while we're in parallel development mode.

kbroch-rivosinc, 2025-05-13:
> I'm not against this idea, nor consistency in general. I suggest we pose this to the sig-documentation list. I'd consider them more appropriate owners for language-specifics. Ideally, the text in the spec is changed to match while we're in parallel development mode.

https://lists.riscv.org/g/sig-documentation/topic/can_we_define_a_consistent/113083679

kbroch-rivosinc, 2025-05-19:
I didn't check for the other form of description text and missed these: 

```
kbroch@kbroch-mac.local:~/rvi/non-repo-repos/riscv-unified-db on  main via  v22.14.0 via 🐍 v3.13.3 via 💎 v2.6.10
❯ for f in arch/inst/*/*.yaml; do if [ "$(yq '.description[].text' "$f" | tr -d '\n' | grep -c -e '^This instruction')" -gt 0 ]; then echo "$f"; fi; done
arch/inst/Zkn/aes64ks1i.yaml
arch/inst/Zkn/aes64ks2.yaml

kbroch@kbroch-mac.local:~/rvi/non-repo-repos/riscv-unified-db on  main via  v22.14.0 via 🐍 v3.13.3 via 💎 v2.6.10 took 7s
```