riscv/riscv-unified-db #745: simply inst/page output: condense synopsis section

kbroch-rivosinc, 2025-05-07:
Even with appreciating white space between sections, I think we could condense the content of the inst/page so that simple instructions stand a chance of fitting on a page (pdf) or not scrolling (html).

Focus of this feature would be to condense the sysnopsis by:

* remove synopsis header (unneeded as it/overview is implied by content after instruction mnemonic at top)
* use the `inst.long_name` + `inst.description` as the content here 
  * NOTE: blocked by #744

kbroch-rivosinc, 2025-05-07:
Here's before after ex:

[image]

ThinkOpenly, 2025-05-08:
There are issues with appending "synopsis" and "description" attributes (I posted in #744).

It could be that the "Synopsis" _section_ doesn't need it's own heading, and could be just dumped nicely _underneath_ "name":

# addi
Add an immediate to the value in xs1, and store the result in xd

dhower-qc, 2025-05-08:
This update looks great!

AFOliveira, 2025-05-08:
BTW, is this really blocked by #744? May look slighlty weird with the other descriptions, but this is so much smaller effort than the other one, that I wonder if we could do it first.

kbroch-rivosinc, 2025-05-12:
> BTW, is this really blocked by [#744](https://github.com/riscv-software-src/riscv-unified-db/issues/744)? May look slighlty weird with the other descriptions, but this is so much smaller effort than the other one, that I wonder if we could do it first.

I don't disagree that it's not really "blocked" in the sense of something completely failing but more that if I try to make a sentence from "`inst.long_name` `inst.description`" we will go from inconsistent grammar to down right incorrect grammar in some cases. Also what @ThinkOpenly mentioned above/#744

So for now, I'll just remove the synopsis heading as a small improvement.

commit b8004a22 (2025-05-12): feature: condense overview section of inst pg tmpl

just make synopsis implicit by removing header
and putting inst.description at top of page
fixes #745
PR #765 (2025-05-13): feature: condense overview section of inst pg tmpl
just make synopsis implicit by removing header
and putting inst.description at top of page
fixes #745
commit f653cd01 (2025-05-13): feature: condense overview section of inst pg tmpl (#765)

just make synopsis implicit by removing header
and putting inst.description at top of page
fixes #745
closed 2025-05-13 by ThinkOpenly