riscv/riscv-unified-db #767: Add tools to generate non-ISA documents

AFOliveira, 2025-05-14:
In the future, I see UDB as a resource that allows to generate full PRMs, which means that we need to add tools to describe both ISA and non-ISA this. Aditionally, we will need to test them with a non-isa extension at first, I suggest the AIA.

At first, I would just like to see this distinction in file restructuring basically have
/data/
        /isa
       /non-isa

I will be working on this , so this is really a placeholder, to check if someone is interested in the same subject and may help :)

ThinkOpenly, 2025-05-14:
Could you expand the "PRM" initialism for us n00bs? 😉

AFOliveira, 2025-05-14:
Of course I can! I will also expand the clarification to the rest of the issue.

**PRM** stands for **Platform Reference Manual**. It's a comprehensive document that describes the architecture, features, and behavior of a platform, including both **ISA** and **non-ISA** components. In the context of UDB, the goal is to eventually support the generation of full PRMs directly, just like we do with ISA Manuals.

To move toward that goal, we need to:

1. **Add tools and structure** to describe both ISA and non-ISA components.
2. **Test the system** initially using a non-ISA extension. I suggest starting with **AIA (Advanced Interrupt Architecture)** as a test case.

As a first step, I'm proposing a **file restructuring** to reflect this distinction more clearly:

```
/data/
/isa/
/non-isa/
```

This is still a work in progress, and I'm currently working on it. This issue is mainly a placeholder to:
- Track the activity
- See if anyone else is interested in contributing or collaborating on this effort

ThinkOpenly, 2025-05-14:
> As a first step, I'm proposing a **file restructuring** to reflect this distinction more clearly:
> 
> ```
> /data/
> /isa/
> /non-isa/
> ```

What do you see as the purpose for "data" versus the other two? Put differently, what would be in "data" that would not be in the others?

Are these top-level directories, and if so, where would the current top-level directories go?

AFOliveira, 2025-05-15:
> What do you see as the purpose for "data" versus the other two? Put differently, what would be in "data" that would not be in the others?
>
> Are these top-level directories, and if so, where would the [...]

 GIthub does not allow me to indent that, so it looked funky. /isa/ and /non-isa/ should obviously be under /data/. Other tool related stuff should be under /tools/. Then, docs and other types of files could have the their own file. I think this goes inline with Derek's view of monorepo approach, right? If you have views or input please help me to structure this idea, since I feel we are clearly needing a small cleanup refactor.

ThinkOpenly, 2025-05-19:
This is very subjective, but I wonder if we should avoid too much disruption until we know how things start to shake out. It seems that `/data/isa` would be just a renamed `/arch`, which isn't too difficult to do later, and I'm not sure it's needed at this point?

Could the non-ISA data start flowing into a `/non-isa` directory now, just to get the ball rolling?

AFOliveira, 2025-05-19:
> This is very subjective, but I wonder if we should avoid too much disruption until we know how things start to shake out. It seems that `/data/isa` would be just a renamed `/arch`, which isn't too [...]

Definitely can and will as a PoC, I am tinkering around with it now and will provide updates when they exist

dhower-qc, 2025-05-19:
FYI, I have a branch that moves files around into `data/arch`, in addition to other changes like creating proper Ruby gems. I'll accelerate that so you can pick it up.