riscv/riscv-unified-db #779: Need to complete adoption of classified register names

ThinkOpenly, 2025-05-18:
**Current value and location**
Many instruction definition still have generic register names like `rd`, `rs1`, `rs2`.

In assembly syntax:
```
arch/inst/Zicfiss/ssrdp.yaml:assembly: rd
arch/inst/Zilsd/sd.yaml:assembly: rs2, offset(rs1)
arch/inst/Zilsd/ld.yaml:assembly: rd, offset(rs1)
arch/inst/Zfh/fsh.yaml:assembly: xs2, imm12($rs1)
arch/inst/Zfh/flh.yaml:assembly: xd, imm12($rs1)
arch/inst/Zfh/fli.h.yaml:assembly: rd, imm
arch/inst/C/c.addw.yaml:assembly: xd, rs2
arch/inst/C/c.subw.yaml:assembly: xd, rs2
arch/inst/C/c.and.yaml:assembly: xd, rs2
arch/inst/C/c.xor.yaml:assembly: xd, rs2
arch/inst/C/c.add.yaml:assembly: xd, rs2
arch/inst/C/c.or.yaml:assembly: xd, rs2
arch/inst/C/c.sub.yaml:assembly: xd, rs2
arch/inst/F/froundnx.s.yaml:assembly: fd, rs1, rm
```

And in field names (too numerous to list here):
```
$ grep -e '- name:' $(find arch/inst -name '*.yaml') | grep -E 'rd|rs1|rs2' | wc -l
1306
```

**Expected value**
UDB has adopted the convention of a leading `x` for general purpose register indices, `f` for floating-point, `v` for vector.

ThinkOpenly, 2025-05-18:
Somewhat related to #622 and #705.

AFOliveira, 2025-05-26:
Should we also enforce that the enconding section and assembly section have a combination of the same parameters?

ThinkOpenly, 2025-09-18:
> Should we also enforce that the encoding section and assembly section have a combination of the same parameters?

There are some weird ones that would need accommodation.

For example (mismatches):
```
name: cm.push
...
assembly: reg_list, -stack_adj
encoding:
  variables:
    - name: rlist
...     
    - name: spimm
```

and (implicit fields):
```
name: c.fldsp
...
assembly: fd, imm(sp)
encoding:
...
  variables:
    - name: imm
...
    - name: fd
```