riscv/riscv-unified-db #784: Add ISA manual yaml for newly published 20250508

james-ball-qualcomm, 2025-05-20:
From Bill T:

At just over one year since the last published set of RISC-V Unprivileged and Privileged ISA specifications we are pleased to announce the publication of a new set of documents as of May 8, 2025. The release is tagged in the riscv-isa-manual repository with 20250508.

This release includes many additions as well as the removal of any unratified content. The full release notes can be found [here](https://github.com/riscv/riscv-isa-manual/releases/tag/20250508). For quick access to the documents in PDF form, see [riscv.org](https://riscv.org/specifications/ratified/).

kbroch-rivosinc, 2025-05-20:
I'll have a go at this unless @dhower-qc really wants it.

dhower-qc, 2025-05-20:
Let's do this on top of #552 since that changes how manuals are represented. We can either do it in that PR, or get the PR merged first.