riscv/riscv-unified-db #788: Missing Fields in `misa` Register Definition

AFOliveira, 2025-05-20:
**Observed:**

The current definition of the `misa` register in our system includes the following fields:
`A`, `B`, `C`, `D`, `F`, `G`, `H`, `I`, `M`, `S`, `U`, `V`

**Expected :**

According to the RISC-V Privileged Specification (Volume II), the `misa` register should define bits for a broader range of extensions. The table below (from the spec) illustrates the standard bit assignments:

[image]

**Gap Analysis:**

Comparing our current implementation against the specification, the following fields appear to be missing from our `misa` definition:
`E`, `J`, `K`, `L`, `N`, `O`, `P`, `Q`, `R`, `T`, `W`, `X`, `Y`, `Z`

**Next steps:**

We should review and update our `misa` register definition to include these standard fields

dhower-qc, 2025-05-20:
Most of those are intentionally omitted since they don't correspond to anything and are read-only-0. (unspecified fields are implicitly read-only-0). The exceptions are:

- E (needs E in UDB)
- Q (needs Q in UDB)

X should never be in the standard definition, and can be added in an overlay.

AFOliveira, 2025-05-20:
Q is in UDB, right?

dhower-qc, 2025-05-20:
ah, so it is

AFOliveira, 2025-05-26:
Fixed by #801

closed 2025-05-26 by AFOliveira