riscv/riscv-unified-db #790: Partially configured documents are breaking

AFOliveira, 2025-05-20:
There seems to be an IDL error that is causing not to be possible to generate partially configured documentation via
`./do gen:htm[config]`. I may try to debug but if someone IDL-familiar would take a look I would appreciate it a lot.

log:

Using singularity environment
Running with 1 job(s)
/home/afonso/riscv-unified-db/.home/.venv/bin/python3 lib/yaml_resolver.py merge arch /does/not/exist gen/arch/rv32
Merging arch: 100%|################| 1640/1640 [00:00<00:00, 10122.13it/s]
[INFO] Merged architecture files written to gen/arch/rv32
/home/afonso/riscv-unified-db/.home/.venv/bin/python3 lib/yaml_resolver.py resolve gen/arch/rv32 gen/resolved_arch/rv32
Resolving arch: 100%|#################| 1640/1640 [00:17<00:00, 94.31it/s]
[INFO] Resolved architecture files written to gen/resolved_arch/rv32
/home/afonso/riscv-unified-db/lib/cfg_arch.rb:653:in `transitive_implemented_instructions': transitive_implemented_instructions is only defined for fully configured systems (ArgumentError)

      raise ArgumentError, "transitive_implemented_instructions is only defined for fully configured systems"
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

dhower-qc, 2025-05-20:
That backend assumes it's working with a full configuration. We'll have to modify the templates to get it to work with a partial config.

AFOliveira, 2025-05-21:
I'll give it a try in fixing it