riscv/riscv-unified-db #796: .vx RVV instructions refer to the GPR source as both xs1 and rs1

camel-cdr, 2025-05-21:
I noticed that the .vx RVV instructions refer to the GPR source as both xs1 and rs1:

https://github.com/riscv-software-src/riscv-unified-db/blob/1c982a4d0d0dc2324b074c36b68dce0dc643f65c/arch/inst/V/vadd.vx.yaml#L10-L18

This might be fine and don't look wrong on the produced artifacts, but I'm not sure it's intended.

dhower-qc, 2025-05-21:
That's a bug -- they should be the same. @AFOliveira, is this already covered by the issue David Brash raised?

AFOliveira, 2025-05-21:
Not really, but I think #676 will fix it whenever we get to it, right?

dhower-qc, 2025-05-21:
Yes, it should.

camel-cdr, 2025-05-21:
I'm trying to use UDB to draft a proposal, should I just use rs1 for both, or are deeper changes needed?

On that note, how would you encode sub-extensions that restrict the supported SEW?
zve32x and similar are currently not supported (https://github.com/riscv-software-src/riscv-unified-db/issues/616), so I don't have anything to go off.

AFOliveira, 2025-05-22:
> I'm trying to use UDB to draft a proposal, should I just use rs1 for both, or are deeper changes needed?

Since those are GPR, default should be xs1. Technically, you can use any of them without the need for deeper changes, literally just change them on that yaml. It would also be a nice contribution :)

To modify pre-existing instructions you have to use `arch_overlay` which as the name suggests allows to overlay existing data. However, if I got it correct, what you need to do is restrict `vtype ` CSR, right? It seems that it is not present in UDB as of now. 

With some AI help I drafted a first approach, but likely there are somethings not entirely correct. If you just want to describe textually the behaviour, you may just chnage the textual descriptionsof each field through `arch_overlay` .
I may be missing some things, @dhower-qc and @ThinkOpenly are definitely more familiar with CSR description in UDB so I'll ask them to correct me if I am misleading in some way.

```
# yaml-language-server: $schema=../../schemas/csr_schema.json
$schema: "csr_schema.json#"
kind: csr
name: vtype
long_name: Vector Type Register
address: 0xC21 # Standard CSR address for vtype
writable: false # vtype is read-only; updated by vsetvl/vsetivli instructions, not direct CSR writes
priv_mode: U # Accessible in U/S/M modes; vsetvl is also U/S/M accessible
length: XLEN
description: |
  The vtype CSR provides the default type used to interpret the contents of the vector register file.
  It is updated by vsetvl and vsetivli instructions.
  The vector type determines the organization of elements in each vector register,
  how multiple vector registers are grouped (LMUL), the selected element width (SEW),
  and how masked-off (vma) and tail elements (vta) are handled.
  The vill bit indicates if the current type setting is illegal.
definedBy: V # Defined by the Vector Extension
fields:
  vill:
    location_rv32: 31
    location_rv64: 63
    description: |
      *Vector Illegal Type*

      Set to 1 by hardware if the combination of vsew and vlmul set by a vsetvl/vsetivli
      instruction is unsupported by the implementation, or if an attempt is made to set
      a reserved encoding for vsew or vlmul.
      If vill is 1, executing any vector instruction (other than vsetvl/vsetivli)
      or attempting to read/write vstart, vxrm, or vxsat CSRs causes an illegal instruction exception.
    type: RO # Read-Only; value determined by hardware after vsetvl/vsetivli
    reset_value: 1 # Typically 1 at reset, indicating an uninitialized/illegal vector type
  reserved:
    location_rv32: 30-8
    location_rv64: 62-8
    description: |
      *Reserved Bits*

      These bits are reserved for future standard extensions.
      Software should ignore values read from these bits and preserve them on writes
      (though vtype is not directly writable, this applies to the conceptual model).
      Hardware should return 0 when these bits are read if not implemented.
      Implementations must ensure that writing non-zero values via vsetvl/vsetivli
      to any implemented portion of these reserved bits results in vill being set.
    type: RO # Read-Only, should read as 0 if not implemented for specific extensions
    reset_value: 0
  vma:
    location: 7
    description: |
      *Vector Mask Agnostic (vma)*

      Controls whether masked-off elements in a destination vector register group are
      undisturbed (0) or overwritten with 1s (1) by masked vector instructions.
      - 0: Masked-off elements are undisturbed.
      - 1: Masked-off elements are overwritten with all 1s.
    type: RO # Read-Only; value reflects state set by vsetvl/vsetivli
    reset_value: 0
  vta:
    location: 6
    description: |
      *Vector Tail Agnostic (vta)*

      Controls whether elements in a destination vector register group past the current
      vector length (vl) are undisturbed (0) or overwritten with 1s (1).
      This applies to both masked and unmasked vector instructions.
      - 0: Tail elements are undisturbed.
      - 1: Tail elements are overwritten with all 1s.
    type: RO # Read-Only; value reflects state set by vsetvl/vsetivli
    reset_value: 0
  vsew:
    location: 5-3
    description: |
      *Vector Selected Element Width (vsew[2:0])*

      Sets the dynamic selected element width (SEW) in bits.
      Encodings:
      - 000: SEW = 8-bit
      - 001: SEW = 16-bit
      - 010: SEW = 32-bit
      - 011: SEW = 64-bit
      - 100 to 111: Reserved (setting these values will cause vill=1)
    type: RO # Read-Only; value reflects state set by vsetvl/vsetivli
    reset_value: 0 # Part of an illegal configuration if vill=1 at reset
  vlmul:
    location: 2-0
    description: |
      *Vector Register Group Multiplier (vlmul[2:0])*

      Sets the vector length multiplier (LMUL).
      Encodings:
      - 000: LMUL = 1
      - 001: LMUL = 2
      - 010: LMUL = 4
      - 011: LMUL = 8
      - 100: Reserved (setting this value will cause vill=1)
      - 101: LMUL = 1/8 (f8, fractional)
      - 110: LMUL = 1/4 (f4, fractional)
      - 111: LMUL = 1/2 (f2, fractional)
      Fractional LMUL values (f2, f4, f8) are only valid if SEW <= (VLEN / (LMUL_denominator * ELEN_min_for_vector_ops)).
      Typically, ELEN_min_for_vector_ops is 8. So for f8, SEW must be VLEN/64 or smaller.
    type: RO # Read-Only; value reflects state set by vsetvl/vsetivli
    reset_value: 0 # Part of an illegal configuration if vill=1 at reset

# Note on reset_value for the vtype CSR as a whole:
# Implementations typically reset vtype such that vill=1, indicating an illegal/uninitialized
# vector type until a vsetvl/vsetivli instruction configures it.
# For example, on RV32, this could be 0x80000000. On RV64, 0x8000000000000000.
# The individual field reset_values above are consistent with this, resulting in vill=1
# and other active fields (vma, vta, vsew, vlmul) being 0.

```

ThinkOpenly, 2025-05-22:
Note: there's a `vtype.yaml` emerging in #544. Indeed, some of the differences should be fed back into that PR, unless this one takes on a life of its own.

ThinkOpenly, 2025-06-24:
The original issue reported has been fixed. Any objections to closing this issue?

If we want to open a new issue for missing `vtype` CSR, perhaps?