riscv/riscv-unified-db #807: add pattern regex to inst_schema.assembly keyword

kbroch-rivosinc, 2025-05-22:
Currently it is simply a string value.  I'm wondering if we could specify something more that might be helpful?

Out of 1151 instructions there's 116 string variations:

```
kbroch@kbroch-mac.local:~/rvi/non-repo-repos/riscv-unified-db on  main via  v22.14.0 via 🐍 v3.13.3 via 💎 v2.6.10
❯ ls -1 arch/inst/*/*.yaml | wc -l
    1151

kbroch@kbroch-mac.local:~/rvi/non-repo-repos/riscv-unified-db on  main via  v22.14.0 via 🐍 v3.13.3 via 💎 v2.6.10
❯ yq '.assembly' arch/inst/*/*.yaml | sort -u | wc -l
     116
```

Initially could be something like `^\w+(?:, \w+)?(?:, \w+)?$` but might be able to make it more specific?

dhower-qc, 2025-05-23:
A few comments/observations:

- We can reduce duplication/potential for error by moving the assembly syntax to the new sub-type definitions (#686)
- We could (should?) at least check that the variables in the assembly string correspond to defined variables in the encoding. That will have to be done through Ruby (there is already a validate function for instructions that checks other properties of the encoding).
- On moving to something more structured: we might find it's required in the end. For example, in some vector instructions the 'vm' field is either present _or absent_ in the assembly based on an encoding value. We can't currently represent that.

AFOliveira, 2025-05-27:
> * We could (should?) at least check that the variables in the assembly string correspond to defined variables in the encoding. That will have to be done through Ruby (there is already a validate [...]

As you mentioned, I think we should be careful doing this because then how can we identify edge cases like the (imm)? Also, there is a subset of Vector extension that ends in ".vxm" that has v0 in assembly and v0 is not really a variable in hardware description of variables. Doing some Regex seems like a reasonable step, but IMO the CI checks should be external, maybe checking agaisnt LLVM or Sail

dhower-qc, 2025-05-27:
I agree with @kbroch-rivosinc, we should add a regex that captures what we have. If it needs to be changed later to be more inclusive, that seems fine.

AFOliveira, 2025-05-27:
@ThinkOpenly 
> Could you elaborate on that?

I am talking mostly about matching variables to assembly field. I think there are some cases, such as the ones you recently commited in which the match is not 1:1, right?

> I'm kinda hoping LLVM becomes a downstream project.
> I'm working on some Sail-based validation, so stay tuned.

I'm waiting! I think in the long term both Sail and LLVM could be downstream users :) 

@dhower-qc to be clear: I think we should do the regex so that IDEs warn ASAP, its just the CI step that @kbroch-rivosinc stated that I think would be great to check against external sources.