riscv/riscv-unified-db #848: Remove "requires" extension dependency and just use "implies"

james-ball-qualcomm, 2025-06-16:
Met with Derek and Dave Weaver and discussed how RISC-V models extension dependencies (in ISA manuals, Profiles&Bases&Extensions spreadsheet, in UDB). We decided to simplify UDB and just have one way of expressing dependencies ("implies") instead of the current two ways ("implies" or "requires"). There are no use cases (current or imagined future) that can justify the extra complexity of have 2 ways to express dependencies.

This will require me to modify the extension YAML schema, update just a few extension YAML files that use "requires" (e.g., S extension requires U extension), update the Ruby code dependent on having both dependency mechanisms (will be simplifying code), and then update ISA Explorer extension format to only have one dependency column.

james-ball-qualcomm, 2025-06-16:
@dlweaver and @dhower-qc

james-ball-qualcomm, 2025-06-16:
@AFOliveira

dhower-qc, 2025-06-18:
For more context: UDB has the two dependency mechanisms because there seems to be two different ways dependencies are specified in the manual and because the profile spreadsheet started by Mark Himmelstein had different columns for the dependency types.

However, we concluded that there was wide confusion (including among the Profile SIG Chairs) about what the difference actually was and no clear reason why the difference existed.

Thus, the change. We can discuss this the UDB SIG meeting today as well.

dhower-qc, 2025-06-18:
@ThinkOpenly

closed 2025-06-23 by james-ball-qualcomm

james-ball-qualcomm, 2025-06-25:
We just decided in the UDB meeting to have one keyword.