riscv/riscv-unified-db #893: Generated C header missing instructions with new subtype schema

jordancarlin, 2025-07-12:
**Describe the bug**
All instructions that have been migrated to use the new type/subtype encoding schema are skipped by `generator.py`, which is used to produce both the c_header and golang outputs. The resulting files are missing both the `MATCH` and `MASK` clauses for these instructions.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `./do gen:c_header`

**Text output/Error Messages**
```
⬢ [podman] ❯ ./do gen:c_header
Using devcontainer environment
Running with 1 job(s)
/workspaces/riscv-unified-db/.home/.venv/bin/python3 /workspaces/riscv-unified-db/backends/generators/c_header/generate_encoding.py --inst-dir=/workspaces/riscv-unified-db/gen/resolved_spec/_/inst --csr-dir=/workspaces/riscv-unified-db/gen/resolved_spec/_/csr --ext-dir=/workspaces/riscv-unified-db/gen/resolved_spec/_/ext --output=/workspaces/riscv-unified-db/gen/c_header/encoding.out.h --include-all
INFO:: Loading instructions from /workspaces/riscv-unified-db/gen/resolved_spec/_/inst
INFO:: Searching for instruction files in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst for target architecture BOTH
ERROR:: Missing 'encoding' field in instruction orn in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/B/orn.yaml
ERROR:: Missing 'encoding' field in instruction andn in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/B/andn.yaml
ERROR:: Missing 'encoding' field in instruction ror in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/B/ror.yaml
ERROR:: Missing 'encoding' field in instruction clmulh in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/B/clmulh.yaml
ERROR:: Missing 'encoding' field in instruction xnor in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/B/xnor.yaml
ERROR:: Missing 'encoding' field in instruction rorw in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/B/rorw.yaml
ERROR:: Missing 'encoding' field in instruction rolw in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/B/rolw.yaml
ERROR:: Missing 'encoding' field in instruction clmul in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/B/clmul.yaml
ERROR:: Missing 'encoding' field in instruction rol in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/B/rol.yaml
ERROR:: Missing 'encoding' field in instruction add.uw in /workspaces/riscv-unified-db/gen/resolved_spec/_/inst/Zba/add.uw.yaml
INFO:: Found 1150 instruction definitions in 1150 files
INFO:: Filtered out 10 instructions due to encoding issues
```

**Additional context**
`generator.py` is looking specifically for an `encoding` field, which no longer exists in these instructions. It needs a new way to retrieve the encoding.

There should probably be golden reference files for these artifacts (similar to what is already done for the instruction appendix) to prevent inadvertent issues like these. Or at least a check in CI for error messages.

omsuneri, 2025-07-15:
assign me please @jordancarlin

ThinkOpenly, 2025-09-02:
> I understand that but it's best to not leave parts of UDB broken just because we may need to fix it in the future, right?

I certainly have no issue fixing things that are broken, especially if the fix isn't a huge effort and there is a reasonably urgent need. I just wanted to note that whatever fix is implemented will very likely need to change again. For the audience here, that probably did not need to be said. :-) For a larger audience, maybe.