riscv/riscv-unified-db #915: Add a test on backends to ensure schema changes don't break them,

AFOliveira, 2025-07-21:
**Is your feature request related to a problem? Please describe.**
Sometimes, schema changes create bugs in the backends as information layout changes and assumptions made on what information looks like are broken. 

**Describe the solution you'd like**
We should aim to add **breaking** statements when this errors happen, instead of allowing silent exits or wrong output generation. Somehow the backends need to identify that not being able to fetch certain information is not an accident and exit with an error. This is more of a problem in backends/generators.

**Describe alternatives you've considered**
We can follow the "mock DB" approach, and create certain artifacts just for testing, but seems overkill to what we are trying to achieve.

AFOliveira, 2025-07-21:
Hello @Sukuna0007Abhi. Please only take new issues once you are waiting for review. This is non-trivial and will require thought process on how to implement. So although it is not hard, it needs to be strategic.

Sukuna0007Abhi, 2025-07-21:
Ok sir I am going for that issue only focusing on that issue then this. @AFOliveira

Shehrozkashif, 2025-07-24:
@AFOliveira 
To ensure schema changes in the database are caught early and handled explicitly, I'm planning the following:

1. **Introduce a `validate_schema(entry)` function** within the generator(s) (e.g., `generate_docs.py`) to verify the presence and structure of required fields before accessing them.
2. Use `try/except` blocks or assertions with clear error messages, so that generators **fail fast and explicitly** when expected fields are missing. For example:

   ```python
   assert "version" in entry.get("attributes", {}), "Schema change detected: 'attributes.version' missing"
   ```
3. Integrate this validation step at the beginning of each generator’s execution flow to ensure consistency across backends.
4. **Add CI coverage** by including a test that runs the generators against mock or real DB entries, and fails when schema expectations are violated.

---

This approach aims to:

* Prevent silent failures due to structural changes in the DB.
* Make assumptions about the schema explicit and maintainable.
* Improve robustness of backend generators and CI reliability.

Let me know if this direction looks good or if there are any changes you'd recommend.