riscv/riscv-unified-db #1026: Bug: Svadu ADUE field type logic is backwards in environment configuration CSRs

Sukuna0007Abhi, 2025-08-25:
## Bug Description
The type() function for ADUE fields in environment configuration CSRs has backwards logic.

## Current (Incorrect) Logic
yaml
```
return (implemented?(ExtensionName::Svadu)) ? CsrFieldType::RO : CsrFieldType::RW;
```

**Rationale**
When Svadu extension IS implemented → ADUE field should be Read-Write (software can control it)
When Svadu extension is NOT implemented → ADUE field should be Read-Only-0 (hardwired to 0)

**Reference**
Identified by sir @ThinkOpenly in PR review feedback.

ThinkOpenly, 2025-08-25:
reference: https://github.com/riscv-software-src/riscv-unified-db/pull/1019#discussion_r2294795371

Sukuna0007Abhi, 2025-08-25:
Yes sir

commit 265dabb3 (2025-08-25): fix: correct Svadu ADUE field type logic in environment CSRs (#1027)

The ADUE field type() logic was backwards in all environment
configuration CSRs.

Fixed logic:
- When Svadu IS implemented  ADUE field is Read-Write (software control)
- When Svadu is NOT implemented  ADUE field is Read-Only-0 (hardwired)

Closes #1026
PR #1027 (2025-08-25): fix: correct Svadu ADUE field type logic in environment CSRs
The ADUE field type() logic was backwards in all environment configuration CSRs.

Fixed logic:
- When Svadu IS implemented  ADUE field is Read-Write (software control)
- When Svadu is NOT implemented  ADUE field is Read-Only-0 (hardwired)

Closes #1026
closed 2025-08-25 by ThinkOpenly