riscv/riscv-unified-db #1041: `henvcfg` upper-half fields need `base: 64`

ThinkOpenly, 2025-08-26:
**Current value and location**
`spec/std/isa/csr/H/henvcfg.yaml`

**Expected value**
All fields in the "upper-half" of `henvcfg` need to be restricted to XLEN 64 (by adding `base: 64`). When XLEN == 32, these fields are not present in `henvcfg` and must be accessed via `henvcfgh`.

(It's probably worth checking other CSRs which have "upper-half" semantics as well.)

ThinkOpenly, 2025-08-26:
The same treatment is required for `menvcfg`.

ThinkOpenly, 2025-08-26:
And `mstatus`

Sukuna0007Abhi, 2025-08-26:
yes sir and I also think hstateen and mstateen

Sukuna0007Abhi, 2025-08-26:
> And `mstatus`
Actually, sir I think, we already handled mstatus correctly in Phase 2 PR #1019 . Both the MDT@42 and SDT@41 fields already have base: 64

ThinkOpenly, 2025-08-26:
I think I'm confusing myself, so apologies...

`menvcfg` and `henvcfg` are always a 64-bits. Their fields should _not_ have any "base:" attributes.
`mstatus` is already correct.

Sukuna0007Abhi, 2025-08-26:
oh ok sir