riscv/riscv-unified-db #1069: Not able to generate CPP hart for example_rv64_with_overlay config

ShashankVM, 2025-09-08:
**Describe the bug**
CPP hart generation fails for example_rv64Z_with_overlay_config

**To Reproduce**
Steps to reproduce the behavior:
1. Run `./do gen:cpp_hart CONFIG=example_rv64_with_overlay`

**Text output/Error Messages**
./do gen:cpp_hart CONFIG=example_rv64_with_overlay

**Screenshots**

[image]

**Additional context**
I'm trying to generate a CPP hart with floating-point extension code so I could validate the floating-point code.

dhower-qc, 2025-09-08:
@henrikg-qc, FYI

dhower-qc, 2025-09-08:
Is there something in the overlay you need? Have you tried with any other config?

ShashankVM, 2025-09-09:
Hi @dhower-qc, I tried without overlay, with the following config I called "rv64F" and I still get the same error.
```
# yaml-language-server: $schema=../schemas/config_schema.json
---
$schema: config_schema.json#
kind: architecture configuration
type: fully configured
name: rv64F
description: An example fully-specified RV64 system
implemented_extensions:
  - [A, "2.1.0"]
  - [B, "1.0.0"]
  - [C, "2.0.0"]
  - [D, "2.2.0"]
  - [F, "2.2.0"]
  - [I, "2.1.0"]
  - [H, "1.0.0"]
  - [M, "2.0.0"]
  - [S, "1.12.0"]
  - [Sm, "1.12.0"]
  - [Smhpm, "1.12.0"]
  - [Smpmp, "1.12.0"]
  - [U, "1.0.0"]
  - [V, "1.0.0"]
  - [Zicntr, "2.0.0"]
  - [Zicsr, "2.0.0"]
  - [Zihpm, "2.0.0"]
  - [Smaia, "1.0.0"]
  - [Smcdeleg, "1.0.0"]
  - [Smcntrpmf, "1.0.0"]
  - [Sscofpmf, "1.0.0"]
  - [Ssaia, "1.0.0"]
  - [Ssccfg, "1.0.0"]
  - [Sstc, "1.0.0"]
  - [Sv39, "1.12.0"]
  - [Sv48, "1.12.0"]
  - [Zicboz, "1.0.0"]
  - [Zicbom, "1.0.0"]

params:
  MXLEN: 64

  # name of the configuration
  NAME: rv64F

  # vendor-specific architecture ID in marchid
  ARCH_ID: 0x1000000000000000

  # vendor-specific implementation ID in mimpid
  IMP_ID: 0x0

  # JEDEC Vendor ID bank
  VENDOR_ID_BANK: 0x0

  # JEDEC Vendor ID offset
  VENDOR_ID_OFFSET: 0x0

  # whether or not the implementation supports misaligned loads and stores in main memory (not including atomics)
  # must be true when Zicclsm is supported
  MISALIGNED_LDST: true

  MISALIGNED_LDST_EXCEPTION_PRIORITY: high

  MISALIGNED_MAX_ATOMICITY_GRANULE_SIZE: 0

  MISALIGNED_SPLIT_STRATEGY: by_byte

  # whether or not the implementation supports misaligned atomics
  MISALIGNED_AMO: false

[... 479 lines cut ...]
  MSTATUS_FS_LEGAL_VALUES: [0, 1, 2, 3]
  MSTATUS_FS_WRITABLE: true
  MSTATUS_TVM_IMPLEMENTED: true
  HW_MSTATUS_FS_DIRTY_UPDATE: precise
  MSTATUS_VS_WRITABLE: true
  MSTATUS_VS_LEGAL_VALUES: [0, 1, 2, 3]
  HW_MSTATUS_VS_DIRTY_UPDATE: precise
  FORCE_UPGRADE_CBO_INVAL_TO_FLUSH: true
  REPORT_GPA_IN_HTVAL_ON_GUEST_PAGE_FAULT: true
  VSTVEC_MODE_DIRECT: true
  VSTVEC_MODE_VECTORED: true
  TRAP_ON_ECALL_FROM_VS: true
  TRAP_ON_ECALL_FROM_S: true
  TRAP_ON_ECALL_FROM_M: true
  TRAP_ON_ECALL_FROM_U: true
  TRAP_ON_EBREAK: true
  TRAP_ON_SFENCE_VMA_WHEN_SATP_MODE_IS_READ_ONLY: false
  MSTATEEN_ENVCFG_TYPE: rw
  HSTATEEN_ENVCFG_TYPE: rw
  PRECISE_SYNCHRONOUS_EXCEPTIONS: true
```

**Screenshot**

[image]

ShashankVM, 2025-09-09:
I tried replacing 64 with 32 in the config file, and then got this error
```
Validating arch: 100%|#################################################################################################################################################################################| 1730/1730 [00:10<00:00, 172.17it/s]
[INFO] Resolved architecture files written to /workspaces/riscv-unified-db/gen/resolved_spec/rv32F
[INFO] 2025-09-09 05:15:37: Allocating more SymbolTables
/workspaces/riscv-unified-db/tools/ruby-gems/udb/lib/udb/obj/instruction.rb:1087:in `encoding': rolw is not defined in 32 (RuntimeError)
```

henrikg-qc, 2025-11-05:
@ayosher also experienced similar issue but halting on rolw with CONFIG=qc_iu.

`riscv-unified-db/tools/ruby-gems/udb/lib/udb/obj/instruction.rb:1087:in 'encoding': rolw is not defined in 32 (RuntimeError)`