riscv/riscv-unified-db #1082: Recent PR violates RISC-V style guidelines (instruction names must be in lower-case)

james-ball-qualcomm, 2025-09-13:
PR #1079 shouldn't have been approved because RISC-V Instruction requires all instruction names to be lower-case.

Please see https://github.com/riscv/docs-dev-guide/blob/main/src/style-guidelines.adoc

[image]

Paul, please ensure you are aware of the RISC-V style guidelines in the above repo if you are going to review these kinds of PRs and approve them.

Abdullah, please make your changes consistent with the RISC-V style guidelines (mostly changing upper-case instruction names to lower-case) and re-submit this PR or create a new one.

ShashankVM, 2025-09-14:
While we are at this, how about creating tests for style rules validation that runs before PRs get merged?

As this project has a huge number of contributors, I think such kinds tests would be useful.

ThinkOpenly, 2025-09-15:
> I guess that the ISA not following all of its own guidelines is one of the reasons for UDB to exist. However, I have not checked this particular case.

I'm just trying to avoid a fire drill 🚒 if there is no 🔥 . Literally, almost every reference to an instruction name in the ISA manual is in upper case, with the exception of code examples.

ShashankVM, 2025-09-15:
> I'm just trying to avoid a fire drill 🚒 if there is no 🔥 . Literally, almost every reference to an instruction name in the ISA manual is in upper case, with the exception of code examples.

I think it's a clear contradiction, Paul. I've submitted my PRs with lowercase instruction names and those have been approved. We can't have inconsistent styles here,  can we?

ThinkOpenly, 2025-09-15:
We need to follow the lead of the ISA manual. Most of the content in question comes directly from that.

There are efforts underway to reverse that flow, but we're not there yet. And, indeed, once that effort gets going in earnest, it is very likely that the content here will be overwritten.

Put another way, I'm not yet convinced that the prosaic content here matters that much, and certainly not more than the ISA content. That day may be coming, but I am not aware of any plan or schedule.

The quality of my reviews of that content probably reflects the level of effort I feel is currently warranted.

james-ball-qualcomm, 2025-09-15:
Example of the name being changed in the other PR.

[image]

ThinkOpenly, 2025-09-15:
We've bifurcated the discussion. Let's continue only here.

Does the change to using upper case break something? Upper case is more closely aligned with the content in the ISA manual. Unless something is broken, I'm not terribly inclined to fix it in UDB first... it should be fixed first in the ISA manual.

ShashankVM, 2025-09-15:
A Pythonic solution, keep a dict of all instruction names, parse the text and check for capitalized instruction names and replace them. AI might be like getting a gun to shoot a mosquito!

ThinkOpenly, 2025-09-15:
I completely agree with the goal of consistent documentation. I claim that UDB is not the best place to fix the issue right now. It's not even clear what the right fix is, given that the ISA manual difference from its own style guidelines.

closed 2025-09-18 by ThinkOpenly